ramapcsx2/gbs-control #533: Slow sync with external clock generator

TheRetroCarrot, 2024-02-09:
I'm having an issue where horizontal sync takes a while to stabilize initially or when changing resolutions.
[https://youtu.be/nJ5TXbfRE94](url)

This is with an external clock gen, at first I thought it might be the generator itself, so I purchased an adafruit pcb with the same results. This never causes an issue after initial sync so long as the resolution doesn't change, image is perfectly stable with no tearing. Output in the GUI is as follows when enabling my preset:

"clock gen reset: 81000000
Active FrameTime Lock enabled, adjusting external clock gen frequency

preset applied: 720x480 (custom) for EDTV 60Hz

source Hz: 59.93941 new out: 59.93946 clock: 80987928 (-12072)"

Disabling FTL causes things to stabilize much faster, but not instantly. This occurs on many different systems all running component cables with VGA output to an HD CRT with native RGBHV.

Anybody have any ideas or suggestions? I've gone over all my connections many times, there is no bridging and everything has a solid connection. If this is normal behavior I'm fine with it, but I don't believe it is?

TheRetroCarrot, 2024-02-09:
So it appears disabling syncwatcher after a system has synced fixed this even on resolution change, the issue is anytime I switch systems I need to re-enable syncwatcher then disable it. Any ideas?

YoImLee, 2024-02-09:
Have you tried a different display? To me it seems like your monitor is sensitive to the algorithm used to bring the input and output frequencies in sync with each other.

ramapcsx2, 2024-02-09:
Yea, I'm sorry but some displays really hate this or that algorithm, in all sorts of variations. I'm surprised that even a CRT does it though. Tough luck :/