ramapcsx2/gbs-control #534: 1024p@50Hz not displayed correctly

awaken1ng, 2024-02-12:
<details>
<summary>Basically, this</summary>

[image]
</details>

1024p@50Hz is the only resolution that does this. Others are nice and crisp comparatively. 

<details>
<summary>720p@50Hz for example</summary>

[image]
</details>

Curiously, 1024p@60Hz is fine. Tested on a couple of monitors (which outputs just like picture above) and a TV (which refused to output 1024p@50Hz due to "unsupported video mode").

Clock gen is installed, but I've tested GBSC for a bit prior to installing it, and it still did that. The board is GBS-8200 V4.0, dated 20231130.

The device is PS2, as could be guessed from the photos, connected with component cable. Display is connected with VGA cable.
Running latest firmware at the time of writing (5293343b2c253a35e85cc5b638431cfaab3ed813).

awaken1ng, 2024-02-13:
Oh, I'm a dum-dum, for some reason thought cycling SDRAM was just, like, resetting the memory.

Setting SDRAM clock to 129 Mhz (according to debug output) specifically makes the problem go away, the higher clocks (144 Mhz and higher) starts showing a lot of "white dots noise" like the stock GBS did, and lower clocks is glitchy like the photo above.

Looks like PAL60 and NTSC seem to both run 129 Mhz in 1024/960p, and PAL50 doesn't, hence the glitchiness.

Presets seem to remember the SDRAM clock, so guess can just that as a workaround for setting the 1024/960p resolution.

ramapcsx2, 2024-02-14:
Hm, back when I coded the SDRAM section, all my GBS boards worked well with them, and I have a lot.. :p
I suppose they still make them a bit different in the wild. Yours seems to not like some particular detail..

Traeskie, 2024-06-30:
> Oh, I'm a dum-dum, for some reason thought cycling SDRAM was just, like, resetting the memory.
> 
> Setting SDRAM clock to 129 Mhz (according to debug output) specifically makes the problem go away, the higher clocks (144 Mhz and higher) starts showing a lot of "white dots noise" like the stock GBS did, and lower clocks is glitchy like the photo above.
> 
> Looks like PAL60 and NTSC seem to both run 129 Mhz in 1024/960p, and PAL50 doesn't, hence the glitchiness.
> 
> Presets seem to remember the SDRAM clock, so guess can just use that as a workaround for setting the 1024/960p resolution.

Thanks for sharing this!

I have the same issue:
Flickering dots with 50Hz 1024p/960p, other modes are fine.
Tried with several different PSU:s up to 2A, no difference.
Setting SDRAM to 129MHz fixes it, all other frequencies have similar issues.
...and I also assumed Cycle DRAM was some kind of reset :D

My board is GBS 8200 V4.0 20240417 and I've only tested with PS2 as input.

ahavasi12, 2025-01-03:
Same issue. Cycle sdram saves the day

A-Dunstan, 2025-10-26:
I have a bit of extra info that might explain why this affects some boards and not others.
I've got a GBS 8200 that has "V4.0 20250416" printed on the PCB. The fitted SDRAM chip is a Hynix HY57V643220DLTP **-7**. Per the datasheet, this chip is only rated for 143MHz.
It has a lot of speckles at 162MHz and higher. Not so much at 144MHz but still the odd speck here and there. It works fine (as expected) at the 129.6MHz setting. Usually SDRAM chips support at least _some_ overclocking so I'm surprised that 144 doesn't work - maybe some of the other SDRAM timings could be loosened to make it work at that speed.
But either way, the problem is likely these slower chips being put on GBS-8200 boards when they're originally designed for the **-6** part that is rated for 166MHz, which would explain why the factory firmware uses 162MHz (the clone makers don't seem to realize that overclocking the slower RAM causes visual issues).