ramapcsx2/gbs-control #536: New PlatformIO build - WiFi issues

jerzyjurgiel, 2024-02-21:
Hi!
I recently upgraded to the new build using platformio. So much better than Arduino studio.
Although ever since new  version, I had constant issues with GUI. Frontend is unable to connect using websockets to the ESP8266. It keeps calling onconnect and onclose right away.
Sometime I can toggle a setting after clicking few times, but for ex. Console Output is always blank.
Earlier arduino build had no such issue and there has not been any HW change on my side. Never happened before.
Any advice?

Metaln00b, 2024-02-21:
> Hi!
> I recently upgraded to the new build using platformio. So much better than Arduino studio.
> Although ever since new  version, I had constant issues with GUI. Frontend is unable to connect using [...]

https://github.com/ramapcsx2/gbs-control/issues/531

jerzyjurgiel, 2024-02-21:
Thanks, I looked through issues but didn't flag that one as similar issue. I guess it would be great to disable OLED support on a build or I'll just use rama's suggestion of modyfing the code :D

jerzyjurgiel, 2024-02-21:
I also just found USE_NEW_OLED_MENU flag. I'll test it later today.

Metaln00b, 2024-02-21:
If you want it would also be possible to place another ifdef statement around the oled code. Then the code is not compiled.

I and maybe a few other experienced people would have to see what the Arduino IDE uses for compiler flags and then add them to PlatformIO. Presumably the heap problems will then disappear. Unfortunately, the problem has only existed since the new OLED menu, although it is not necessarily a programming error.

I hope that my assumptions are related to your problem. The opening and closing of the sockets in the developer tools of the browser only appear if the debug messages for the console have been activated via the debug settings in the gbsc interface and I assume that this is normal. It's just not normal that you don't see any output 😅

ramapcsx2, 2024-02-21:
This heap mem check is dodgy to begin with. I guess it's trying to create new websockets for loosing connection with the old one (due to the radio issues). Maybe those can be recycled instead, dunno..
But it should be safe to play with the watermark value for free heap, to get the rest working as normal.