ramapcsx2/gbs-control #537: Fatal error: Waiting for packet header

Ehh300, 2024-02-22:
I went to the wiki to setup esp8266 and it errors.

Arduino: 1.8.11 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Executable segment sizes:

IROM   : 662864          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 29988   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 2352  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 6068  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 35280 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 701272 bytes (67%) of program storage space. Maximum is 1044464 bytes.
Global variables use 43700 bytes (53%) of dynamic memory, leaving 38220 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:53:de:12
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Traceback (most recent call last):
  File "C:\Users\mine\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/mine/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2916, in main
    esp.change_baud(args.baud)
  File "C:/Users/mine/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "C:/Users/mine/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 705, in change_baud
    self.command(self.ESP_CHANGE_BAUDRATE, struct.pack('<II', baud, second_arg))
  File "C:/Users/mine/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 347, in command
    p = self.read()
  File "C:/Users/mine/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 292, in read
    return next(self._slip_reader)
  File "C:/Users/mine/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2045, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ramapcsx2, 2024-02-22:
This is a generic USB upload error, but I can't tell what causes it. It might be a bad USB cable or bad USB hub etc... It is not related to anything software though. The uploader times out waiting for the right ESP response.

Ehh300, 2024-02-22:
I'm guessing the esp8266 is bad then. Both usb port and cable work. Thanks for looking.

ramapcsx2, 2024-02-22:
"Crystal is 26MHz" are we sure about that? Check the Arduino IDE uploader settings, whether there's some options about that. It might be 40MHz.

Ehh300, 2024-02-22:
How do i do that. Never done this kind of thing.

Ehh300, 2024-02-22:
Just tried on my laptop now. Got this

. Variables and constants in RAM (global, static), used 44740 / 80192 bytes (55%)
    SEGMENT BYTES DESCRIPTION
     DATA 2604 initialized variables
     RODATA 6024 constants
      BSS 36112 zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR) ,used 62531 / 65536 bytes (95%)
     SEGMENT BYTES DESCRIPTION
      ICACHE 32768 reserved space for flash instruction cache
      IRAM 29763 code in IRAM
. Code in flash (default, ICACHE_FLASH_ATTR) ,used 667936 / 1048576 bytes (63%)
      SEGMENT BYTES DECRIPTION
        IROM 667936 code in flash
esptool.py v3.0
Serial port COM6
Connecting...
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC:
Uploading stub...
Running stub...
Changing baud rate to 460800
A Fatal esptool.py error occurred: Timed out waiting for packet header

Then in blue bar it says Done uploading

Metaln00b, 2024-02-22:
Try reducing the upload baudrate to 115200 otherwise it could be an usb to serial driver error.

Ehh300, 2024-02-22:
What about the crystal at 26 Mhz?

Ehh300, 2024-02-22:
[image]

Metaln00b, 2024-02-22:
> What about the crystal at 26 Mhz?

[image]

Ehh300, 2024-02-22:
I changed com port from 6 to 8 and got a different error.
Timed out waiting for packet headerFeatures: WiFi

ramapcsx2, 2024-02-23:
I'd love to tell you exactly what's wrong, but I haven't worked with Arduino for a long time now.
There is a problem with the uploader, where it most likely is set to an incompatible setting.

Please check with the instructions here, and try to set everything as shown:
https://github.com/ramapcsx2/gbs-control/wiki/Software-Setup#esp8266-ide-settings