ramapcsx2/gbs-control #538: WIFI settings not saved

fkooistra, 2024-02-22:
Hi, I upgraded to the new build using Arduino studio without any errors. Although everything seems to be functioning as expected, when I select a home network with its (correct) password the settings are not saved after reboot and the device starts in AP-mode again. Could this be build related?

Metaln00b, 2024-02-27:
Try using 2.4GHz WiFi

diegoapn, 2024-02-27:
firmware problem, i compiled today and didn´t work, i used other firmware and worked at first time.

Metaln00b, 2024-02-27:
Would one of you be so kind and look into the serial console to see if an error message or something similar appears after setting the credentials and post it here? 

Maybe something will appear there.

At the same time, it would also be great if you could find the relevant line of code and analyze it if necessary, that would help.

Thanks a lot

YoImLee, 2024-02-28:
Verify your flash size settings in the Arduino IDE and make sure you are running esp8266 version 2.6.3, not 3.0 or newer.

fkooistra, 2024-02-28:
I downgraded the esp8266 boards package back to 2.6.3, and used the following library versions:
- ESP Async Webserver: 1.2.3
- ESP8266 and ESP32 OLED Driver for SSD1306: 4.4.1
- ESP8266-ping: 2.0.1
Basically, following the software setup instructions and ignoring library updates suggested by the Arduino IDE. This seems to work and my GBSp-control connects to my Wifi-network.

fkooistra, 2024-02-28:
Issue fixed by using exact package and library versions as provided with setup instructions.

closed 2024-02-28 by fkooistra