ramapcsx2/gbs-control #539: Sega Saturn image shaking

NightKid2k, 2024-02-25:
Hello, greetings from germany :)

i am searching for a resolution in terms of my issue with an gbsc AIO.
Using the clear one like in the image below
[image]
I have a pal sega saturn in combination with the gbsc (updated to the latest FW from here through arduino IDE).
Picture everything looks amazing good, but the image is shaking in the saturn menu and in some movie ingame scenes.
https://www.youtube.com/shorts/rCAhoEfO_2M
Funny thing is, in 3D game play there is no shaking at all.

I tried RGB scart with sync on composite video, RGB via component with sync on Luma, but the issue remains.
Changed a lot of settings in the webUI including syncwatcher and so on, brought no change.

Connecting the saturn directly to the TV without the gbsc is working without any problems.
Also the shaking becomes more less the more the system is running in combination with the gbsc.
Other consoles like the mega drive (genesis) or NES are having 0 issues.
I am pretty at my end with ideas for the moment.
Thanks in advance

NightKid2k, 2024-02-29:
Additional info:
restarting the saturn sometimes makes the image shaking much less.

ramapcsx2, 2024-02-29:
This is intermittent sync loss. The CSync extraction works most of the time, but has trouble locking on to the signal fully.
It might be caused by cabling issues or there's a problem in the sync input termination on the GBS.

NightKid2k, 2024-03-01:
i saw people are talking about putting some resistors on the sync line.
could this maybe help in my case?

ramapcsx2, 2024-03-01:
That's exactly it, yea. I don't know how these various custom GBS are built though. The vanilla pcb can be modified with a resistor towards ground on sync.

ramapcsx2, 2024-03-01:
Experiment! :p 
The recommendation was smth like 100Ohms on the vanilla board.
Yours may have that, or maybe there is even a sync stripper (look for a small chip marked like LM1881 or so), in which case the resistor wouldn't work nor should be needed... It really depends on the hw you have.