ramapcsx2/gbs-control #542: Clarification about the power supply bypass capacitors

solidhit2, 2024-03-10:
I just finished modding my GBS-8200 (v5.2) by assembling information from various sources. My device works great and I cannot believe someone made this for free and who know how many contributed any way the could. Just one question. Instead of following the official suggestions for the power supply bypass capacitor mod, I changed the electrolytic capacitors paralleled with C23, C41, C42, C48 (three 100uf/16v and one 22uf/25v) with 220uf/16v. I didn't parallel any ceramic at 10uf or 22uf with the original ceramic capacitors. My system is solid and I don't see any artifacts / waves / noise. Am I wrong, could this cause problems along the way?

ramapcsx2, 2024-03-10:
Thanks for the nice words man :p
So the whole capacitor thing is a bit esoteric. I experimented with that to fight wavy noise in the image, and it sometimes does work, more or less, but it's not inconclusive.
The main indicator that they help is that the power rails look less noisy on the oscilloscope after adding them.
For this, ceramics work much better when the area is close to a consumer (like the big ASIC). 
Electrolytes help as bulk storage more nearby the power regulators, they keep the line more stable overall.

I think you can't really do this very wrong, and if your image looks okay, then all is well :)

solidhit2, 2024-03-10:
Thank you for the quick response. The reason I made this question is that I know ceramics filter staff electrolytics cannot. Furthermore, the proximity to the ASIC is very important, I agree. I have four ceramics at 22uf (I don't think they make larger capacitance SMD ceramics). I'm gonna install them parallel with the original ceramics. I also don't think anything bizarre is going to occur. I have to say though, that's a lot of capacitance near that ASIC!

ramapcsx2, 2024-03-11:
Iirc the large value worked great at the most noisy frequency domain, but dunno, my scope and experience back then weren't the best xD

solidhit2, 2024-03-11:
Just for completion, I'm letting you know I went in and installed the four 22uf ceramics at their proper location. I really can't see the slightest difference when I connect my PS2 (SCART RGB 480p) or my Amstrad CPC (SCART RGB 576i). Needless to say I'm not going to revert any change, no need to. At least, making these mods gives me peace of mind that I've done all I could to improve ... frankly what was already good.

ramapcsx2, 2024-03-12:
There's a lot of variance in scaler systems, where analog meets digital, and the PCB quality isn't the best regarding compliance, shielding from noise etc.
There will be source systems with a frame (field) rate that hits exactly at a weak spot in the spectrum, producing visible waves in the output image.
This will be different or masked with any number of random settings, but lowpass video filters and such affect it the most.

What the extra capacitors do is, they narrow the range of these affected frequencies, so they're like an added safety to never hit any :)

solidhit2, 2024-03-12:
You consciously made a choice to improve a device which was built to do a job for as little money as possible. The fact that there is an analog aspect to the operation of this board, only makes things harder. People criticizing the project, do it as if the board was designed by rama. It was not and it's not just one board. The original design has been copied and revised multiple times. Some designs take into consideration, the analog needs of the board, some don't. Surely, along the way you find that certain things can be improved by swapping components but there are limits. The fact that the project is a vast improvement on such a messy situation is to me, admirable.

closed 2024-03-18 by solidhit2