ramapcsx2/gbs-control #543: Jumpy image with SNES and Yoshi's island

KonaKonaX3, 2024-03-19:
Hello,

Just finished to built the a GBS-C last weekend.

This unit has all the recommended Wiki mods performed .

- Clock generator installed.
- No 1881N separator circuit, as most of my RGB cables are Csync routed wether it is from the system or with a separator.
- Im using a 2A 5V Xiaomi AC adapter, 5V power is steady as I could measure during idle or load states.
- wemos//nodemcu is powered with the 5V in not the 3v3.

The unit was tested these days with a PAL megadrive switched to 60hz with a Csync RGB scart cable, played and finished games without a single issue.

I started testing with a PAL SNES (1chip with switchless at 60hz) connected with RetroGamingCables RGB Scart (I believe it uses a Csync separator circuit on the scart head) .

I notice that when the SNES is connected with that cable the unit looses WiFi connection, this doesnt happen with the MegaDrive/Genesis (weird)

But anyway I proceed to play,

Played through Super Ghouls and Ghosts without an issue.

Tested ActRaiser , no issue.

But with Yoshi's island something happens as the screen  remains shaky during stage title displays and when running a bit into a stage.

Steps to reproduce the issue the fastest possible.

Go to 1-4 (first castle stage) the screen jumps a bit while displaying the name of the stage.

When the initial screen with Yoshi at the catle its shown the shake stops and picture stays great.

Then:

Run right until the first pseudo 3D trap door, after the second coin bucket screen starts to shake like it is vibrating forever.

Tested so far without results:

Going to default presets.

Changing every one of them while on the shaky screen part.

Moving the picture.. nothing changed.

Still more SNES games to test, does yoshi's island have any weird use of resolution that is different?

sidneivl, 2024-06-22:
I'm facing same problem playing Super Mario 2 in Mario All starts

https://youtu.be/-CM12qhUpvU

Do you have a fix for this issue?

Regards

igor1019, 2024-08-15:
Same here. Im using snes 1chip p3 with scart on luma and there is shaking image only in yoshis island. Is there any fix for that? 

Demo video:

https://youtu.be/guRtgv3Ky3E?si=SqN9hHOXMZ9-tGA-

igor1019, 2024-08-19:
What motherboard revision is your super nintendo?
Mine is 1-chip-03 and Im having the same problem
I think that is the Csync ou sync on luma go through scart cable
When I connect by AV cable directly, the shaky image stops.

TonyRyodan, 2025-01-07:
Hello same here, also with final fight :C 

Any solutions?

TonyRyodan, 2025-01-07:
> I'm facing same problem playing Super Mario 2 in Mario All starts
> 
> https://youtu.be/-CM12qhUpvU
> 
> Do you have a fix for this issue?
> 
> Regards

In your case, I belive pressing ''RGBS Sync'' button will solve

AchillesPDX, 2025-02-23:
Anyone discover a solution to this? Same exact issue here with Yoshi's Island as well as with my Super Game Boy. Running an NTSC SNES Jr. with the "3-wire" mod to enable RGB output. I'm then running RGBs+ground to the RGBs+ground pins on the GBS.

I've added the 100 ohm resistor between the Sync pin and ground and have also replaced C33 and C35 with 1nF capacitors.

My N64 and all my other SNES games work beautifully, but Yoshi's Island and my Super Game Boy all bounce up and down very rapidly.