ramapcsx2/gbs-control #544: How do I flash new firmware onto the GBS-C using Windows 11

Route66Fan, 2024-03-20:
The GBS-C that I have is the one by Mr. Ao. 
I would like to do this, but I can't easily access the micro USB port on the WiFi expansion board due to it being located behind some RCA\Component jacks.

YoImLee, 2024-03-20:
If your computer and GBS are on the same network, you can try the "enable OTA" option on the GBS. It should pop up as a device in your Arduino IDE just like it would when it's connected through USB. If it doesn't work you may have to set your firewall network profile to "private" in your Windows settings, or whitelist the Arduino IDE in there somewhere. That might help.

Route66Fan, 2024-03-21:
> If your computer and GBS are on the same network, you can try the "enable OTA" option on the GBS. It should pop up as a device in your Arduino IDE just like it would when it's connected through USB. [...]

That worked, thank you!

closed 2024-03-21 by Route66Fan

popperclopper, 2025-05-21:
Hi. Were you able to reflash GBS? Does sync work?
Do RCA and S-video work?
What firmware did you install?