ramapcsx2/gbs-control #545: Connecting an old PC to the VGA port on a GBS-control sometimes results in garbled video

Route66Fan, 2024-03-22:
The GBS-control that I have is the Mr. Ao version. 

I think that the problem may be that the GBS-control is detecting the VGA signal as a 70hz signal instead of a 59hz, or 60hz, signal. 

I can sometimes temporarily fix the problem by going into the "Reset/Restore" menu on the GBS-C & selecting "Reset GBS". 

What could be causing this problem? [image]

[image]

[image]

[image]

mgomersbach, 2024-10-26:
What value would that be?

Saf98, 2024-10-29:
> What value would that be?

I think its the scaling settings with the +/- buttons.