ramapcsx2/gbs-control #548: Red & Green horizontal lines in just one quadrant of the the display signal.

multigamesystem, 2024-04-16:
Im getting a really weird issue with my GBS control setup when hooked up to my Neo Geo MVS.  I doubled checked that it wasnt the LCD or some weird dead pixels.  Its only showing up with the Neo Geo MVS I didnt notice it when running other arcade hardware with the GBS Control.  Ive tried another MVS board and tested without the cart (as seen here)  Its driving me crazy cause the image looks really good otherwise.

[image]
[image]

ramapcsx2, 2024-04-16:
Try a different preset maybe, or try the Web UI options that move the picture. It looks like memory garbage that is meant to be masked but that's not working, something that can happen with GBSC.

multigamesystem, 2024-04-17:
> Try a different preset maybe, or try the Web UI options that move the picture. It looks like memory garbage that is meant to be masked but that's not working, something that can happen with GBSC.

I havent tried any of the physical mods to my GBS could there possibly be an interference issue going on?  Changing presets is really clunky using the UI is there a way to do that over FTP or something?