ramapcsx2/gbs-control #553: OLED screen - Line Filter toogle

golimonkey, 2024-05-23:
I often toggle line filter depending on what console I connect to GBSC and changeing profiles with OLED menu works good but linefilter setting doesnt change. I tried enableing Save filtering per slot option but it doesnt work with oled menu. Could someone add linefilter toggle in OLED menu somewhere? I tried to do it myself but cant understand the code properly.

Sorry if request is inappropriate.

ramapcsx2, 2024-05-23:
Hey ho, it's appropriate, of course. 
I don't have time for it, but this request serves perfectly to show @way5 how these feature requests formed the messy preset system :p
Since he's currently overhauling it, maybe he can fit this in.

way5, 2024-05-25:
@ramapcsx2 yeah, I've already seen that fuzzy logic. I am doing some testing, after changing way too many things. There is no way someone do something with it without changing everything :)

@golimonkey I can not promise for sure :) but I'll try to do something similar to what you've mentioned.

way5, 2024-06-03:
Hey @golimonkey !
Just FYI, line filter control seems to work properly in 'pio' branch through WebUI. So if you're comfortable with uploading the new firmware onto your GBSC, I'd recommend you to try. 😉 
See the How To in README.

golimonkey, 2024-06-03:
@way5 Ty for your info, I will try 'pio' branch! 
But I'll still be looking forward for your implementation on OLED menu.

way5, 2024-06-11:
Hi @golimonkey 
I hasten to inform there is an update you requested in `pio` branch. I expect to have a feedback from you on that update.  😉

golimonkey, 2024-06-15:
@way5 Great news. I'll get back to you. Flashing it today.

golimonkey, 2024-06-15:
@way5 Hi. Tried it just now but cant compile pio branch. I get errors with missing heather files, tried to fix it by editing paths but didnt manage. Not enough time and very little expirience working on other peoples projects...

Error: sketch\src/wserial.h:16:30: fatal error: WebSocketsServer.h: No such file or directory

 #include <WebSocketsServer.h>

It may very well be that its my fault. I downloaded zip(gbs-control-pio), renamed it to gbs-control and tried to compile with arduinoIDE. That works with main branch for me.

Let me know if there is something that I'm missing, probably project isn't set up correctly.

ramapcsx2, 2024-06-15:
@golimonkey The branch is being developed using [PlatformIO](https://de.wikipedia.org/wiki/PlatformIO), which is a more advanced environment than pure Arduino. If something doesn't work, you may have to install that.
Toolchains and SDKs man, it's the coder's life :p

golimonkey, 2024-06-15:
Tnx @ramapcsx2, it try in in VM then.

ramapcsx2, 2024-06-15:
No promises though, but it's worth a try :p

way5, 2024-06-15:
> @way5 Hi. Tried it just now but cant compile pio branch. I get errors with missing heather files, tried to fix it by editing paths but didnt manage. Not enough time and very little expirience [...]

Please read README.md in "pio" branch. This says you need to install all required libraries and remove main.cpp if you compile using Arduino IDE.

golimonkey, 2024-06-15:
@way5 Ty for instructions. I think I'll be able to do it now. 

I have two diy GBSC-s so I'll keep pio branch on one for testing.

[image]