ramapcsx2/gbs-control #555: Video Tearing/Artifacts

diggles4, 2024-06-09:
I just completed this build per Voultar's guide and installed the optional Si5351. I am using a VGA to HDMI adapter, similar to his in the video.

Web GUI works great, but I have not been able to get an image on any display. Most displays give "no signal" but I was able to capture an intermittent image on a single display (iPad via Orion): https://i.imgur.com/0IQFPrH.png

This is the output I get with 480p. With 1280x960 I can get more visual artifacts, while the other modes produce no image at all. Pass Through makes only audio come through.

Looking for advice one where to start diagnosing. Thank you!

ramapcsx2, 2024-06-09:
Hey, there is some kind of big problem, either in the input path, or the Si5351 enabled output path.
To check out whether it is the Si, you can just disconnect one of the cables to it, either SCL or SDA are fine.
This makes the software think that there's no clockgen installed, and use the normal mode.

diggles4, 2024-06-09:
Sure enough, disconnecting SDA cleaned up the signal. I have tried resoldering each of the connections on Si5351 but still have the same issue. Is it possible that the Si5351 is dead?

ramapcsx2, 2024-06-09:
Bad Si5351 is a possibility, but it could also be a bad connection of the clock output (into the ASIC), or any other such issue.

diggles4, 2024-06-09:
Thank you for the help, rama, much appreciated! I will order another Si5351, as my joints are all show conductivity to ship pins without bridging.

As for the PSU... any chance that's the culprit? The reason I ask is because I have to plug it in a specific sequence for it to power the board and ESP. Seems if I plug into outlet before barrel, it almost never powers both boards. Barrel before outlet, leads to higher success rate. If sequenced properly, all LEDs act appropriately.

ramapcsx2, 2024-06-09:
Well, that is also weird :p
I can't really debug your board from here, but if something seems fishy, go investigate! :)

diggles4, 2024-06-11:
Well, as an update, I've now tried the following and still have this issue.
-Re-soldering ESP and Si5351
-Tried a second Si5351
-Reflashed ESP just because

One of the only things I've been reluctant to go back and check are the add-on capacitors... I assume this couldn't possibly be the issue though?

Also, I figured out the PSU thing. If I have the VGA->HDMI adapter powered by the 5v spot near the jack, I have inconsistencies with ESP booting properly when plugging in. If I don't have the adapter powered by the board, or plug it in second, I have no issues.

ramapcsx2, 2024-06-12:
There's an HDMI dongle? This might be the cause for the clockgen sync issue. Some of those don't like slightly off spec clocks.
If you can, try to use the VGA out directly, if only to confirm that it works then > HDMI dongle is bad.

YoImLee, 2024-06-12:
Did you check the description in Voultar's video? If you followed along exactly with the video there's a chance you soldered the 100 Ohm resistor for the RGB sync input incorrectly, connecting it to the RGB blue input. I'm not sure if this would affect the component input but it's worth checking anyway.

diggles4, 2024-06-14:
I can confirm it does not work with VGA only either, just stays as a black screen with the monitor detecting an input but not able to connect. 

For what it's worth, I've still only been able to get a "picture" from the Orion app on iPad, which is a bit bizarre. Other TVs/monitors just stay blank or claim "incompatible format".

I also tried a different PSU (WSU050-4000 by Triad). Same results all around.

Could these 10 micro farad capacitors could have any effect? I'm hesitant to mess with them. These are the types I installed and they _seem_ to be fine: https://www.digikey.com/en/products/detail/samsung-electro-mechanics/CL21A106KPCLQNC/3888060

I have not checked the 100ohm resistor, as I understand that only effects the VGA (or RGB) input.

Thanks again for the assistance, everyone :)

ramapcsx2, 2024-06-15:
Hey, I think we've exhausted the easy things to suggest :p 
Something isn't right when your clock gen should provide the clock to the output section, but it works when the ASIC does it itself.
What that something is.. unless you oscope some signals, it's hard to know.

diggles4, 2024-06-26:
Well, the plot thickens! I figured I'd pick up another ESP and GBS for the heck of it (since I already had two PSUs and Si5351). I have now assembled both and have the same issue on both!

Is it possible that the capacitors I used for C23, C41, C42, and C48 (linked above) are incorrect? Should I have not removed C11 on a GBS8200 board (black buttons)? Does any of the prep work I've done for future SCART (removing pots and adding 100 ohm resistor) have anything to do with the component input issues I've had?

YoImLee, 2024-06-26:
Can you show us some pictures?

diggles4, 2024-06-26:
[image]
[image]

Please ignore my terrible soldering job... I promise I'm getting better.

YoImLee, 2024-06-27:
Looks okay to me, what about the resistor on the back?

diggles4, 2024-06-27:
Does this resistor have anything to do with component input though?

[image]

YoImLee, 2024-06-27:
I'd say there's a chance it could mess things up, try removing it and see if the component input starts working. If it does and you want to still be able to plug in RGBs sources down the line mount the resistor like this instead (image from Voultar's video description):

[image]

diggles4, 2024-06-27:
Wow, thank you for pointing that out! I completely missed that comment on his video.

I have now fixed that resistor - no improvement to component.

YoImLee, 2024-06-27:
You could try enabling developer mode in the web UI and taking a look at the log when turning on the system.

diggles4, 2024-06-27:
I enabled developer mode. Here is what I get when switching to 1080p output (no image):
source Hz: 59.93802 new out: 66.53770 clock: 79402704 (-28597296)
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
give up

Here is what I get when switching to 480p (distorted image - see original pic):
sync skipped ofr wrong: 88.49
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
retry
ABHT: large diff
give up

For the sake of being completely thorough, this is my BOM:

ESP: https://www.amazon.com/dp/B010O1G1ES
Si5351: https://www.amazon.com/dp/B07X7ZT7KJ
GBS: https://www.amazon.com/dp/B01IBJCAIG
PSU: https://www.digikey.com/en/products/detail/triad-magnetics/WSU050-4000/3094945
PSU2: https://www.amazon.com/dp/B09W96X88K
Capacitors: https://www.digikey.com/en/products/detail/samsung-electro-mechanics/CL21A106KPCLQNC/3888060
Resistor: https://www.digikey.com/en/products/detail/panasonic-electronic-components/ERJ-3EKF1000V/196074

And as for what I did...

1. Flashed GBS software to ESP
2. Removed RGB potentiometers and replaced with bridge
3. Added resistor (pic above)
4. Removed C11 from GBS
5. Added capacitors, stacked on top of C23, C41, C42, and C48
6. Bridged GBS (jumper) to disable the GBS original controller
7. Attached all wires, ESP, and Si5351 as shown above

diggles4, 2024-06-27:
Yes, the GBS Control appears to work just fine with the External Clock Generator disabled. Similar results when disconnecting the SCL or SDA wire from the Si5351.

Hmm, I guess I should just try a third Si5351 and see... purchasing from a different seller this time. Stay tuned! Thanks again for the help :)

diggles4, 2024-06-29:
Well, the new Si5351s came in (different seller this time: https://www.amazon.com/dp/B09DM96KGS) and they BOTH work perfectly.

Long story short, don't buy these: https://www.amazon.com/dp/B07X7ZT7KJ

Thank you, rama and YolmLee. I appreciate you!!!

closed 2024-06-29 by diggles4