ramapcsx2/gbs-control #558: Deinterlacing not working on Commodore 128

VanessaE, 2024-07-25:
Interlaced video modes on the Commodore 128 are not displayed properly -- the exact failure mode varies a bit, probably depending on the computer's video chip settings (it's quite flexible).

Sometimes, it'll show one field continuously, while the other field just appears momentarily a couple of times a second.  During those intervals where only one field is shown, it flickers like one would expect.

See for example "I-Paint", which you can find here:
https://www.zimmers.net/anonftp/pub/cbm/c128/graphics/Ipaint/ipaint15h_d81.zip )

Other times, both fields will be present consistently, but without the the proper vertical offset (i.e. you effectively get 240 lines instead of 480), and they'll be flickering at the right rate, but not 100% stable.  One field or the other seems to occasionally "hold" for an instant longer than it ought to.  This is the same sort of effect you might get from a retrocomputer emulator on a PC when displaying interlaced video, if the PC can't render the emulator's video stream fast enough.

See for example the well known but blandly-named "VDC demo", which you can get here:
https://csdb.dk/release/?id=44110

I have tried both bob and adaptive deinterlacing modes, neither works so I assume the firmware simply is not detecting that the video is interlaced.

Note:  if you're trying this on a "flat" C128, you'll need the 64 kB video RAM upgrade to run those two programs.  That or use a C128-DCR if you have access to one.  Lacking those, you can also just try any program that can display 50 text lines, or any program that installs an 8x16 font.  Such things don't need the extra video RAM and will run on any C128.

If it matters, the 8200 gets its sync in "composite" form: H and V fed to it through an XOR gate.

VanessaE, 2024-08-27:
No reply after a month?  🙁

Anyway, interlacing just doesn't work at all lately -- that's not new, that it worked at all in the above-mentioned demo was kind of a fluke, I have not since been able to reproduce it.

ramapcsx2, 2024-08-27:
Oh hey, I suppose even the base "240p" resolution will be different enough from others that it's not fully detected.
The deinterlacer stage requires a solid detection on the progressive format to be able to work.
There's nothing you as a user can do, I think. It needs to be worked at from a dev viewpoint.

VanessaE, 2024-08-29:
After some more testing, it seems like not only is deinterlacing completely non-functional, but also if the displayed image strays too far from NTSC sync rates and whatnot, that's when GBS-Control just gives up and displays garbage (along with throwing bad sync at the monitor).

If it displays the image at all, it just overlays one field on top of the other without the required half-raster displacement.

And I do have the "force PAL 50 Hz to 60 Hz" option enabled.