ramapcsx2/gbs-control #559: Getting an error message when compiling: Compilation error: call of overloaded 'getParam(int)' is ambiguous

TurboMastodon, 2024-07-27:
This is the full error message

C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino: In lambda function:
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9545:59: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *p = request->getParam(0);
                                                           ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9545:59: note: candidates are:
In file included from C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:70:0:
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:360:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:362:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:373:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino: In lambda function:
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9564:59: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *p = request->getParam(0);
                                                           ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9564:59: note: candidates are:
In file included from C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:70:0:
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:360:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:362:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:373:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino: In lambda function:
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9626:67: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *slotParam = request->getParam(0);
                                                                   ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9626:67: note: candidates are:
In file included from C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:70:0:
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:360:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:362:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:373:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino: In lambda function:
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9673:72: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *slotIndexParam = request->getParam(0);
                                                                        ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9673:72: note: candidates are:
In file included from C:\Users\ident\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:70:0:
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:360:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:362:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:373:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9681:71: error: invalid conversion from 'const AsyncWebParameter*' to 'AsyncWebParameter*' [-fpermissive]
                 AsyncWebParameter *slotNameParam = request->getParam(1);
                                                                       ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino: In lambda function:
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9711:51: error: call of overloaded 'getParam(int)' is ambiguous
         AsyncWebParameter *p = request->getParam(0);
                                                   ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9711:51: note: candidates are:
[... 4 lines cut ...]
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:362:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:373:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino: In lambda function:
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9806:58: error: call of overloaded 'getParam(int)' is ambiguous
                 request->send(SPIFFS, request->getParam(0)->value(), String(), true);
                                                          ^
C:\Users\user\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9806:58: note: candidates are:
In file included from C:\Users\ident\OneDrive\Documents\Arduino\libraries\gbs-control\gbs-control.ino:70:0:
[... cut ...]
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\user\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:373:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^

exit status 1

Compilation error: call of overloaded 'getParam(int)' is ambiguous

Pr3ttyF1y, 2024-08-08:
Don't update your packages in the IDE. Just download the repository linked in the wiki. I had the same error as I updated my packages from the IDE, but rolling it back and manually extracting the zip from the link to the code in the wiki allows it to compile.

Chandler-Kluser, 2024-12-17:
> Don't update your packages in the IDE. Just download the repository linked in the wiki. I had the same error as I updated my packages from the IDE, but rolling it back and manually extracting the zip from the link to the code in the wiki allows it to compile.

This is not true, there is no static libraries in gbs-control repository to be used as packages. In order to fully compile the esp8266 firmware you must use the packages from the IDE.

The issue is not using IDE to download the packages, but the packages versions theirselves. I was fully able to build the firmware with:

```
Wire at version 1.0
ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays at version 4.6.1
ESP8266WiFi at version 1.0
ESPAsyncTCP at version 1.2.4
ESP_Async_WebServer at version 2.6.0
Hash at version 1.0
DNSServer at version 1.1.1
ESP8266mDNS at version 1.2
ArduinoOTA at version 1.0
```

reminding only to use `esp8266` at version `2.6.3` in boards manager as wiki says.

Chandler-Kluser, 2024-12-23:
Just to point out, I only manage to fully run gbscontrol on esp8266 with CPU at 80MHz and using these libraries:

https://github.com/me-no-dev/ESPAsyncTCP
https://github.com/me-no-dev/ESPAsyncWebServer

cloning them into `/home/$USER/Arduino/libraries` folder