ramapcsx2/gbs-control #560: [Question] Can I Limit Which Resolutions Get Upscaled?

DeadSkullzJr, 2024-07-27:
As the title implies, I want to limit how resolutions are upscaled. Example, if I wanted to upscale 240p to 480i and not 480p, can I do this? Or if want to only upscale 240p signals to 480p? I have a use case where I don't want my existing 480i and 480p input signals to be touched, but due to an incompatibility for 240p signals on a television, I just want to upscale the 240p signals if possible. Also, can I disable the deinterlacing aspects at will or is it always forced on? I do video captures, and I already wrote up my own deinterlacing scripts for post editing, I would rather stick with that if I can.

My questions may raise some eyebrows, but I'm working on a complex setup, and I need some really tuned option capabilities for it. So please, regardless of whether or not it makes sense to you, I only come with questions in hopes of getting answers, nothing more. I appreciate any potential responses.

ramapcsx2, 2024-07-28:
Yo,
240p to 480i is such an uncommon conversion that nothing is prepared for that. I think it is generally possible, but it requires quite weird programming of the ASIC. If such a preset doesn't exist atm, then it doesn't look good.

DeadSkullzJr, 2024-07-28:
> Yo, 240p to 480i is such an uncommon conversion that nothing is prepared for that. I think it is generally possible, but it requires quite weird programming of the ASIC. If such a preset doesn't exist atm, then it doesn't look good.

That's unfortunate, it would make games that use both 240p and 480i depending on the aspects sort of be seamless if 240p was made to be 480i, similar in fashion to what my television does to my 240p composite signals. Only downside with the 480i upscaled signals there is it sort of removes flickering in places where things are supposed to flicker, but instead leaves it solid with combing.

Main intention for this though was for 240p component signals due to my television nor my capture device accepting 240p YPbPr signals.

ramapcsx2, 2024-07-28:
It will be certainly better to have 240p > 480p than any interlacing methods. What you had in mind would likely fall apart because of one reason or another. Unless you specifically need to have 480i, and you know you do, it's not going to be good :p

For 480i sources, such as the PS2, the ASIC has motion adaptive deinterlacing. You get the best effort solution a scaler can do, so it will look okay'ish, with about 1 frame of lag more.

DeadSkullzJr, 2024-07-28:
> It will be certainly better to have 240p > 480p than any interlacing methods. What you had in mind would likely fall apart because of one reason or another. Unless you specifically need to have [...]

240p to 480p upscaling works great if the games I am playing are always 240p all the time, which is the case for some of my games, however certain games for later systems obviously swap between 240p and 480i. Upscaling those 240p portions to 480p would fix that portion of the problem, but then there is still 480i. Thing is, I do captures as the same time, and quite frankly I am not the most pleased with the deinterlacing methods these devices pack, considering I created my own custom post edit script to deinterlace my interlaced captures anyways, I am completely fine with 480i. My idea was to make it so only the 240p resolution is targeted, and upscaled to 480p for games that are only 240p, and 480i for games that swap between 240p and 480i. If I force 240p to 480i for games that run both resolutions, I get a constant 480i feed, and because of that, my television can deinterlace that by itself without a major drawback, and my capture footage can be deinterlaced with my custom script in post.

ramapcsx2, 2024-07-28:
Maybe what you want is to switch the deinterlacer to bob, instead of motion adaptive.
With bob, you should be able to post process as needed.

DeadSkullzJr, 2024-07-29:
> Maybe what you want is to switch the deinterlacer to bob, instead of motion adaptive. With bob, you should be able to post process as needed.

I have tested bob deinterlacing before via various other means, among other methods of deinterlacing, most approaches to bob deinterlacing causes a noticeable drop in visual quality of the overall contents being viewed (usually makes the overall visual quality fuzzier), while I have no doubts that the implementation here is likely better than how most other solutions have the the method integrated, there is the inevitable issues with bobbing effects and such during recordings, not to mention the occasional moments where some of the frames still have combing and such. Processing something that half baked in a recording makes the situation far more complicated. The thing about these basic methods used, they are dependent on external means to hide away some of the flaws, a television's chipset for example will automatically enhance the visual characteristics of the contents being viewed, thus the viewer won't notice as much glaring issues if at all, however stick something like a monitor or a capture card in the mix, the flaws are much more apparent because neither solution usually has chipsets and such to spruce up the content visuals. While for most people this is fine, and sure it absolutely is since you still get a progressive feed by the end of it that isn't entirely bad, but for me personally, I strive to get the best quality captures when possible, and considering I worked up my own means of deinterlacing my capture feeds, and have the computational power to push it along, all I need really is a method that gives me proper control of the resolutions, 240p in this case, so I not only can view the contents on my television under a compatible resolution (480i and or 480p), but I can capture it as-is as well and avoid the deinterlace drawbacks of these methods altogether in favor of my post editing method.

Just a wild thought, probably doesn't sound the most practical but... Would the idea to integrate a process that upscales 240p to 480p, then interlace the 480p signals to 480i work better in this case? My television has me curious on the concept of upscaling 240p to 480i over composite (and s-video if it had a port). Considering how fast the process is handled from upscaling 240p to 480p, theoretically there shouldn't be a huge hit in latency when interlacing the progressive footage to 480i if such a thing was integrated. It would offer the ability for 240p resolutions to work alongside the already existing 480i segments of games. I know how to downscale 480i to 240p, but the reverse I never quite worked out.

ramapcsx2, 2024-07-29:
Bobbing leaves all the information intact, it just looks flicker'y... I think post processing should be able to fix it