ramapcsx2/gbs-control #561: Gbs board not responding help!!

Mekefux, 2024-08-07:
Hello, last night I received all the components, I made the project, with everything already soldered, when I enter the interface it only appears

"gbs board not responding" and when I restart it tells me to check the power or wiring

I tried changing the slc and sda, check if anything was making contact and nothing, I removed the clock generator to see if that was it and no, I re-did all the soldering and neither, the esp32 when I connect it turns on the blue light and turns off The interface still lets me log in and the red LED on the board turns on. I used this diagram and there's no problem. I can't get it to work. Could some board be damaged?
[image]

Saf98, 2024-08-23:
Did you do all the other mods?

ramapcsx2, 2024-08-23:
Did you set the jumper o puente? :p 
Don't worry if this is your first project like this. There is going to be one mistake that can be fixed. Nothing is broken :)