ramapcsx2/gbs-control #563: White pixels + "GBS Board not responding!"

caden3ds, 2024-08-19:
I installed the wifi chip and clock generator using Voultars tutorial, but whenever I try to change resolution in the first menu, I get "GBS board not responding!". Every other settings seems to work fine, for example when I use the picture control settings I can see it move. I do get a signal but it seems blurry and noisy with white pixels. What should I do?

caden3ds, 2024-08-19:
I set Cycle SDRAM to 129Mhz and now the noise is gone, but I still need help with changing resolutions

caden3ds, 2024-08-19:
Tested for continuity and everything seems fine, does anybody know what would it look like if the GBS8200 wasn't in debug mode? I think that might be it since I might've broken the pads for the debug mode

caden3ds, 2024-08-19:
Reflashed with the correct board instead of the default one but it didn't change anything

caden3ds, 2024-08-19:
Tried to resolder the debug pins and it looks like the ground is 100% broken, is there any other way to put it into debug mode?

ramapcsx2, 2024-08-20:
Hey,
there is no debug mode. Maybe you're thinking of the debug pin signal that has to be soldered?

caden3ds, 2024-08-20:
Yeah this thing at 26:19 https://youtu.be/1AVXhiTlmgo?si=wKLb6qSk_w4Qr3lJ&t=1579

caden3ds, 2024-08-21:
Looking at this page: https://ramapcsx2.github.io/gbs-control/Wiki/Build-the-Hardware.html it seems that its not needed for changing resolutions, and I do see correct timings when I click get video timings, so I guess the debug pin is working

caden3ds, 2024-08-21:
[image]
I soldered it to pin 37 seen here, and it works but now I have no signal and it says "ABHT: large diff" in the console

caden3ds, 2024-08-21:
The error only shows when the clock generator is enabled

caden3ds, 2024-08-21:
Also get "sync skipped ofr wrong: 111.16" and "Error: trying to tune external clock frequency while clock frequency uninitialized!"

caden3ds, 2024-08-22:
Rewired the clock generator and I think its working? Rebooted with clock gen enabled and it seems to be fine

closed 2024-08-22 by caden3ds