ramapcsx2/gbs-control #565: GBS board not responding after reflashing arduino

RPM-Q, 2024-09-05:
Hi, I had an issue where after changing resolution settings I was no longer able to access the control panel, I would get to the splash screen but it would be stuck with the connection error icon in the top right, however I would still get a picture with the settings I had saved to the preset. Reflashing with erasing sketch + wifi settings did not fix that issue, so I tried to reflash by erasing all flash contents. This allowed me to access the control panel again (and obviously I lost my preset but that's not a huge deal), but now no picture comes through and I get the "GBS board not responding" error in the debug log when choosing a resolution option. I know my soldering is correct because before this I had no issues with getting a picture and changing resolution/other settings. The LED on the GBS board comes on and stays on, and the one on my arduino (NodeMCU 1.0) quickly flashes, turns back on for a few seconds, and turns back off, whether or not a picture is being sent to it.
Any help would be greatly appreciated!

ramapcsx2, 2024-09-06:
Hey,
it's been far too long for me to remember the LED behaviour of the 8266 kits, so this doesn't right a bell.
Did you try flashing regularly once more?
Also I don't remember whether / how terminal logging worked on these, but if you can get any logs from the board, that would likely help :)

RPM-Q, 2024-09-06:
Hi,
I reflashed it a few times with a couple different versions of the esp8266 board manager between 2.6.3 and 2.7.4 and with all the settings normal, and still nothing.
This is the whole output log (including trying to set a resolution at the end):
```
startup
(WiFi): still connecting..
! power / i2c lost !
! power / i2c lost !
Please check board power and cabling or restart!
Station disconnected, reason: 2
Station disconnected, reason: 2
Station disconnected, reason: 2
(WiFi): AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
user command f at settings source 4, custom slot 65, status 0
GBS board not responding!
```
I also verified that the component cables I'm using are still working by connecting them directly to a TV and I was able to get a picture like that.

ramapcsx2, 2024-09-06:
Okay, we know the ESP is working fine at least, but I believe it is failing to talk I2C over the assigned pins.
Arduino has an option to select a target devkit, of which there are some variants. Very likely you have one selected where the I2C signals "SDA" and "SCL" are not on the correct pins.

ramapcsx2, 2024-09-06:
Try selecting a different "Board" here:

[image]

RPM-Q, 2024-09-06:
I'm very confused because I know I chose the same board I did when I initially flashed my esp, I even double checked the module and the order details for it to be sure before flashing it this time (and to be extra sure, I checked the connections with a multimeter too and they're still connected and to the right pins). But, I tried many different board options and every one either failed to compile or had the same issue

ramapcsx2, 2024-09-06:
Oh well, the problem space is these 2 signals (SCL, SDA). Check the wiring, maybe check whether you can see the used pins go high/low on an ESP reset, etc.