ramapcsx2/gbs-control #566: PRESETS BUG

domenyX, 2024-09-06:
In the "presets" tab, you created several with the name "custom". Since I didn't create anything. And the ones I had disappeared. There is no longer the "empty" option for me to create a new preset, and below there are also those with strange names. Does anyone know how to fix it?
[image]

maniaco007, 2024-09-06:
acesse a gbs via navegador e use o comando: http://gbscontrol.local/spiffs/format

domenyX, 2024-09-07:
> acesse a gbs via navegador e use o comando: http://gbscontrol.local/spiffs/format

Obrigado! Deu certo =)

darkymtp, 2024-09-29:
Up to date firmware has "remove preset" button.