ramapcsx2/gbs-control #568: Feature Request: Scanlines with 480p sources (PS2, etc) and CRT Filter

blaiddx64, 2024-09-08:
I want to request scanlines support for 480p sources without need to use Bob deinterlace method. Also, if possible, want to request the possibility to add new image filters such as CRT, Smooth, etc like Retrotink does...

Nuck-TH, 2024-09-09:
TiVA is not programmable chip, it is an ASIC. So it can do only what is implemented in silicon. Scanlines are already a hack, and afaik it isn't possible to expand them beyond what is currently implemented.

ramapcsx2, 2024-09-09:
Thanks @Nuck-TH. 
There is so much detail knowledge in this project, I'm not sure it could all be written down even.

But yea, in terms of features, what GBSC offers is already the best this hardware can do, and it only got there with some questionable methods in places ;p

Maybe someone can find other ways to use the digital stages for effects, but it'll never be some amazing slot mask emulation or anything.
I'm just glad the chip had in it the flexibility to do from 240p all the way up to 1080i :)

blaiddx64, 2024-09-09:
Thank you for the clarification!

closed 2024-09-09 by blaiddx64