ramapcsx2/gbs-control #569: Néo Geo sync

pkerga, 2024-09-09:
Hello. I started building my own GBS Control. I made some purchases on AE a few days ago and this weekend, I started. What an incredible satisfaction to see the screen light up. It works pretty well 😎.
I have a problem with my Neo Geo AES, like many. With the GBS Control from Il had, by putting 'Sync on green' on and playing with 'SyncWatcher' in the options, it worked after a while. Not with my build.

[image]

I have several ideas: 
- change capacitors C33 and C35 to improve sync performance.
https://ramapcsx2.github.io/gbs-control/Wiki/Sync-on-Green-Capacitor-Replacements.html

- add a LM1881 on the sync track. 
- trying a RGB to Component converter and run that to the GBS.
- stop playing with my Neo Geo 😥😅😥😅.

What do you think?

ramapcsx2, 2024-09-09:
I think it was more about the logic configuration regarding the sync processor with these MVS.
The sync signal itself should be in good shape, but the combination of odd refresh rate and possibly some tricky serration pulses or such is what makes none of the autodetect methods find proper settings, and so it keeps trying forever to get a good sync lock.
It could be some random setting in segment 5 / sync processor, is my guess.

pkerga, 2024-09-10:
Thanks, what do you mean by "segment 5" ?

ramapcsx2, 2024-09-10:
For developers, this is the register range that contains various sync extractor settings. It's not accessible for users, but has to be implemented in the GBSC code to work.

pkerga, 2024-09-10:
Just to be sure, do you think using a RGB to Component converter could help ?

ramapcsx2, 2024-09-10:
Eh, 60/40 towards not helping. It is about the sync portion, and if such a transcoder would process sync into something better working, then yea. But a bit unlikely.

Chandler-Kluser, 2025-02-04:
> Hello. I started building my own GBS Control. I made some purchases on AE a few days ago and this weekend, I started. What an incredible satisfaction to see the screen light up. It works pretty well [...]

did you try? Does it work?

Chandler-Kluser, 2025-07-25:
Solved [here](https://github.com/ramapcsx2/gbs-control/issues/569#issuecomment-2632873099)

Chandler-Kluser, 2025-07-25:
Solved [here](https://github.com/ramapcsx2/gbs-control/issues/118#issuecomment-2636015597)