ramapcsx2/gbs-control #573: Up button (D7 pin) not working for OLED

owen-rp2a03, 2024-11-04:
I've tried multiple things but cant quite seem to get one of the three buttons for the OLED/OSD to work. I can press down and select, but can't press up for some reason. I checked with a scope and all three are pulled high and do get pulled down by the button, but only up wont work. I personally can't really read the code and am not sure if there is something missing preventing it from working...

owen-rp2a03, 2024-11-04:
Experimented, resolved, please excuse me

flamebox, 2024-11-28:
#define HAVE_BUTTONS 1
Comment out these two lines：
 // attachInterrupt(digitalPinToInterrupt(pin_clk), isrRotaryEncoderRotateForNewMenu, FALLING);
 // attachInterrupt(digitalPinToInterrupt(pin_switch), isrRotaryEncoderPushForNewMenu, FALLING);