ramapcsx2/gbs-control #574: Release Versioning

sirjeannot, 2024-11-16:
Hello,

just received my GBS-C and there is no way to know if the firmware is current or not. There is no versioning anywhere, being in the osd menu or on the webpage. Showing the commit id would help, either in the OSD or on the webpage.

Just an idea. The project doesn't seem to be maintained somehow.

Thanks!

_A new user (mostly for VGA input)_

markusbkk, 2025-01-16:
Did you ever find a way to check the version? I'm having the same issue. On top of that, both GBS-C devices I received (they're McBazel branded ones in an orange plastic shell) have a yellow tint and the blue luma control knob doesn't do anything.

sirjeannot, 2025-01-24:
I did try every last version of the major releases for dependencies causing issues. worked out well, but really not a great sign for this project...

dkozel, 2025-02-04:
Thanks @sirjeannot for pointing out the need to use the v2.x of the Async Webserver. Here's the link to the code at that point for anyone looking: [ESP Async WebServer v2.10.8](https://github.com/ESP32Async/ESPAsyncWebServer/tree/b70d36f2bec270e504bdc739b91a0fcbf3d84377).