ramapcsx2/gbs-control #580: No signal, no connection indicator in webui

cenedek, 2024-12-05:
My diy gbs-c is that i have just built, is not working.

It is a v5.2 board with Nodemcu v2
Checked connections, theyre all fine. Nodemcu gets 5.5v too

There is no video output at all.

It is hard to get into the webui, then it always shows that there is no connection. I can't change any settings, only switching between AP and Wifi mode works. 
The notemcu's blue led tends to light up after some minutes of operating and it stays lit for some time
[image]
.

Any advice?
[image]

Chandler-Kluser, 2024-12-23:
I am facing the same issues here, esp8266 always showing low signal strength (even in not AP mode) with no video signal in any condition. My board is actually the VC9900 (GBS8200 based board).

I also had some trouble to fully build the firmware and make it work, I shared more information in issue #556 

I am worried even if my board is actually dead, because I removed DEBUG jumper to check if it could behave when I bought the board, but it does not output video anymore.

Here are some pictures from my setup:
[image]
[image]

is there an easy way at least to check if the original GBS8200 (variant) is working or if it is something related to the mod done or the firmware?

Thanks in advance!

Chandler-Kluser, 2025-01-09:
As wiki says, low signal strenght the most common cause is the ESP8266 position, which was exactly where I placed 😅

Placing microcontroller far from video chip reduced signal interference.

About black screen, I have just identified the solder had broken so soldering debug pin back to D6 made the setup to work and look FANTASTIC!!