ramapcsx2/gbs-control #587: Shaky image playing some SNES games.

TonyRyodan, 2025-01-07:
My problem is the same as Igor had here: https://github.com/ramapcsx2/gbs-control/issues/562
The majority of games works fine with no problem, but Yoshi Island and Final Fight are unplayble.

CONFIG:

Snes, conected by SCART (Sync on luma)

I aready tried to:
*Change the power suply;
*Change the SCART cable;
*Change the SNES cartrigde;
*Change configs(belive all of them), also the resolution;
*Upgrade the firmware.

I tested it with the OSSC and there is no such problem, :/ also it works fine using an old CRT TV with composite cable.

Please help

AchillesPDX, 2025-02-23:
Same issue here with Yoshi's Island and my Super Game Boy - did you resolve it?

TonyRyodan, 2025-02-24:
> Same issue here with Yoshi's Island and my Super Game Boy - did you resolve it?

Hello, I just had kept the ''RGBS Sync'' button unpressed and the game is playable now, the problem with this ''solution'' is that sometimes the image glitches for a sec, but after that happen it never glitches again, only one time per game bot.

I tried dozens of things, and they all failed, it was exhausting days and I ended up with this temporary solution, I have an OSSC here, and it gives me a better image on the SNES, so I abandoned the GBSC for the SNES, only use that GBSC for PS1, PS2 and others consoles with interlaced images or that change resolution in-game.

AchillesPDX, 2025-02-24:
> > Same issue here with Yoshi's Island and my Super Game Boy - did you resolve it?
> 
> Hello, I just had kept the ''RGBS Sync'' button unpressed and the game is playable now, the problem with this ''solution'' is that sometimes the image glitches for a sec, but after that happen it never glitches again, only one time per game bot.
> 
> I tried dozens of things, and they all failed, it was exhausting days and I ended up with this temporary solution, I have an OSSC here, and it gives me a better image on the SNES, so I abandoned the GBSC for the SNES, only use that GBSC for PS1, PS2 and others consoles with interlaced images or consoles that changes resolution in-game.

You have an RGBS Sync button somewhere? Is that on your board or in the app? I built my own GBS control from the bare board and only have buttons labeled Up, Down, SW, and Menu

[image]

TonyRyodan, 2025-02-24:
> You have an RGBS Sync button somewhere? Is that on your board or in the app? I built my own GBS control from the bare board and only have buttons labeled Up, Down, SW, and Menu
>
> [image]

Oh I see, the one I have is made by RetroScaler

[image]