ramapcsx2/gbs-control #591: cannot compile

Imran-imtiaz48, 2025-01-13:
i cannot compile due to this error 

Compilation error: call of overloaded 'getParam(int)' is ambiguous

AnvilByte, 2025-01-23:
you can try using older versions of the libraries, I was able to compile them when I used the latest versions on January 2nd

AnvilByte, 2025-01-24:
> > you can try using older versions of the libraries, I was able to compile them when I used the latest versions on January 2nd
> > Edit: Actually I ended uo downloading an even older version of the server library, from the same day as the latest commit of this repo and it had less in use glitches for me then the one from January 2nd
> 
> Ho did you downloaded older libraries? I have the same issue and want also try to compile with older libs.

click on the commits list(the button in the top right of the explorer view that looks like a curved arrow around a clock), then find the commit there(you can use the filtering option in the top right) then click the <> button to view the repo as that commit, then just click the green code button then download zip

btw the commits I used is "Update install-arduino-ide.sh" for ESPAsyncTCP and "v2.10.4" for ESPAsyncWebServer, the other 2 haven't been updated in a long time so I just used the latest masters

dkozel, 2025-02-04:
I've just tested with [ESP Async WebServer version 2.10.8](https://github.com/ESP32Async/ESPAsyncWebServer/tree/b70d36f2bec270e504bdc739b91a0fcbf3d84377) and the final commit of [ESP Async TCP](https://github.com/ESP32Async/ESPAsyncTCP).

It'd be slightly nice for GBS Control to update to the version 3 of ESP Async WebServer, but I haven't looked at the changes to the APIs. I asked the developer of the ESP Async TCP code about it's future and they said no new code for the ESP 8266 will be done. There's a different library ([Async TCP](https://github.com/ESP32Async/AsyncTCP)) for the ESP32.

dkozel, 2025-02-04:
I've updated the code to work for ESP Async Webserver 3+ at PR https://github.com/ramapcsx2/gbs-control/pull/596

anteriorsafe, 2025-02-13:
Hi, i had the same issue and found a very simple solution.  install the latest version of ESP Async WebServer, then via arduino, downgrade to version 2.10.8. once installed, removed the folder for the latest version of Async WebServer and hit Verify/Compile. 

It should work after that. and dont forget to turn the GBSC with the knob down.

pkarza, 2025-03-13:
I am a little frustrated - the hardware mods... fine.  I HAVE been able to load this async webserver 2.10.8 - I think it is loaded....  the wiki and the master should be updated to reflect this.  I imagine this will only work when the power is connected to the board, and not the usb?  I am using a esp 8266 lolin wemos d1 r2 and mini.  

When I plug it into the usb the led comes on for a couple of seconds and then goes out.  I can't find any wifi... What am I doing wrong??

Smallmodd, 2025-03-24:
I am having the same issue but with the error "Compilation error: invalid conversion from 'const AsyncWebParameter*' to 'AsyncWebParameter*' [-fpermissive]"

rpf16rj, 2025-05-12:
I have this issue, yet. I compiled recent code using this pack by Mundo Yakara that has Arduino IDE portable and all necessary libraries, just replace the sketch with the recently inside the folder Arduino_IDE_Portable\portable\sketchbook.