ramapcsx2/gbs-control #595: Gen clock feature broken on latest release

Iraito, 2025-01-30:
I tried on 2 different GBS and with 3 different clocks, it never worked, then i downgraded to "gbs-control-96e67732af167fe25966bace8557bd19cf562300" which is like the last release rama worked on and the clock gen works perfectly.

Iraito, 2025-01-31:
Let me update after further testing:
While the older version works, from time to time i get a horizontal line stuck for like 30 to 60 seconds in the middle of the screen, then it vanishes and i get no more lines.

The clock gen feature needs to be fixed and possibly improved.

manadream, 2025-02-04:
I am experiencing the same problem. I tried 3 different clock gens, with the final one being an Adafruit official one, and they all have the same problem:

```
ExternalClockGenerator enabled
user command 5 at settings source 2, custom slot 65, status 4
FTL on
Active FrameTime Lock enabled, disable if display unstable or stays blank! Method: 1 (vtotal only)

Restart
user command a at settings source 2, custom slot 65, status 4
<reset>
Activity detected, input: Component

2345678
Format change: 1 <stable>
preferencesv2.txt opened
loading from preset slot A: /preset_ntsc.A
 (debug) ign. length: 0x61
no stored display clock to use!
ADC offset: R:43 G:40 B:43
clock gen reset: 81000000
Active FrameTime Lock enabled, adjusting external clock gen frequency

preset applied: 720x480 (custom) for NTSC 60Hz 

Error: trying to tune external clock frequency while clock frequency uninitialized!
sync skipped sfr wrong: 0.00
```

And then I get `Error: trying to tune external clock frequency while clock frequency uninitialized!` repeatedly. Changing the switch lock method does not change the outcome.

Iraito, 2025-02-04:
> I am experiencing the same problem. I tried 3 different clock gens, with the final one being an Adafruit official one, and they all have the same problem:
>
> ```
> ExternalClockGenerator enabled
> user [...]

Maybe i found the only working commit, it's called gbs-control-reduce-clockgen-latency, i got the clock gen working and everything is perfectly stable.

manadream, 2025-02-05:
> Maybe i found the only working commit, it's called gbs-control-reduce-clockgen-latency, i got the clock gen working and everything is perfectly stable.

I can't find that commit, or a branch or fork with that name. Can you share a link?

Iraito, 2025-02-05:
https://github.com/ramapcsx2/gbs-control/tree/reduce-clockgen-latency

manadream, 2025-02-05:
I tried all the way back to 2021 and still the same, which makes me think for my case it's an issue with the debug pin connection based on this thread https://github.com/ramapcsx2/gbs-control/issues/241

manadream, 2025-02-05:
Yup, it was a wiring issue, silly me. Works on the latest branch of master for me now. But I was having problems with knockoff clockgens before (which i did wire correctly) and now this Adafruit one is working.

Iraito, 2025-02-05:
In my case the problem was solved only by using an older commit, i think this project needs some refactoring and checking up.

Jameeur, 2025-02-22:
@Iraito could you help me with my case? I have the same issue as well with the latest firmware and you said an older commit fixed this but how did you update your scaler with an older commit? Because I use PyFlasher by using the firmware as a .bin file and flashing the scaler that way. Let me know if you have a .bin file for that specific commit

Iraito, 2025-02-23:
@Jameeur https://www.mediafire.com/file/u4fxilj1h1abfms/arduino-1.8.11.7z/file

Ready to compile

Jameeur, 2025-02-24:
Thanks for the reply @Iraito but I expected a .bin file. Do you know how to make it into a .bin file step by step?