ramapcsx2/gbs-control #608: Unable to flash firmware....

pkarza, 2025-03-13:
I have modified the hardware... No problem...  But I am compile upload to the 8266.  Would someone please list the exact software?  I have the latest Arduino IDE and I have had to change the esp async webserver version to 2.10.8 (I believe).  It will flash at that, but the power on the board led will come on for about 2 seconds and then flashes off.

I think the master might need updating?  I saw an update and I tried it, to but I am dead in the water.  I tried on two of the exact boards.  I am sending them back to Amazon because I think they are lemons.... Or am I doing something wrong?  They were the D1 mini boards

ABelliqueux, 2025-03-13:
I believe it will turn off automatically if there is no video signal. Did you try while feeding a video input ?

pkarza, 2025-03-13:
> I believe it will turn off automatically if there is no video signal (or it doesn't sync). Did you try while feeding a video input ?

Actually, I didn't try it without a video signal.  I have ordered a larger format and will try it.  If this is true, the wiki should be updated with this information.

pkarza, 2025-03-17:
New 8266 and lower driver 2.8.x ...  Working great!

closed 2025-03-17 by pkarza