ramapcsx2/gbs-control #612: GBS Control Not Responding

CB2point0, 2025-04-09:
So, getting a GBS board not responding error, can't figure out where, although this isn't my cleanest soldering, iron has been giving me absolute fits and turning off randomly.

Duponts are temporary, just testing.

Resistors on the back are also temporary, I thought I had 100 ohm but didn't, so I put 10 1k ohm in parallel for the time being while I get an order together.

One of the pads came off the jumper, so I scraped up the trace I believe that goes to it and patched it over temporary. I have checked continuity at R43 and some other points with the jumper mess, but I don't know if that's a good sign or a non sign. Really wanted to make sure everything works before I go and solder in good solutions.

When plugged in, the esp8266 gets power, and I can access the GBS control webpage, but I get no output and a GBS board not responding readout if I change any input settings. I have switched the SDA and SCL lines just in case, zero difference.

Right now, I don't know if I'm missing something, or if the board is the way of the dinosaurs and I just need to buy a new one. It's ugly as hell, and I wouldn't be surprised if it was dead, just what to give it an honest once over before it's called dead.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

CB2p0, 2025-04-14:
I checked through and couldn't find anything. Ordered a 2nd GBS8200 and wired it up with much less fuss, and from what I see it looks like the debug "jumper" I tried to patch in didn't work, so right now it looks dead but it might not be, tbd. I'm going to keep playing around with it once I figure out a different issue I am now encountering. Other issue I think it's sync related, possibly with just my wii and not the GBS Control at all. Don't know if it's my GBS Control, the Wii or the cables. I'll open another issue down the line if I can't figure it out. Quick question though, just so I can be thorough, as I am pretty new to playing around with video signal converters. If I show a colortastic tornado reminiscent of a VCR out of tune but passthrough works perfect, is that most likely to be a sync issue? (I am going to rule out the TV itself asap, it was already not wanting to show input signal from an unmodified gbs8200 while a monitor I have showed input just fine.)

CB2point0, 2025-04-14:
I checked through and couldn't find anything. Ordered a 2nd GBS8200 and wired it up with much less fuss, and from what I see it looks like the debug "jumper" I tried to patch in didn't work, so right now it looks dead but it might not be, tbd. I'm going to keep playing around with it once I figure out a different issue I am now encountering. Other issue I think it's sync related, possibly with just my wii and not the GBS Control at all. Don't know if it's my GBS Control, the Wii or the cables. I'll open another issue down the line if I can't figure it out. Quick question though, just so I can be thorough, as I am pretty new to playing around with video signal converters. If I show a colortastic tornado reminiscent of a VCR out of tune but passthrough works perfect, is that most likely to be a sync issue? (I am going to rule out the TV itself asap, it was already not wanting to show input signal from an unmodified gbs8200 while a monitor I have showed input just fine.)

ramapcsx2, 2025-04-14:
Cheers man, appreciate it! :)

Both units are likely not dead, but there's going to be either shorts or missing / bad connections, or what I said about the SDA/SCL thing.