ramapcsx2/gbs-control #613: Compilation error: call of overloaded 'getParam(int)' is ambiguous

trev1976, 2025-04-24:
Hi

Im trying to put firmware onto my ESP8266 D1 Mini , ive followed the wiki but when i try to verify/compile i get the error : 
Compilation error: call of overloaded 'getParam(int)' is ambiguou

Any idea how i can sort this ? Thanks

Here's the log : 

C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino: In lambda function:
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9545:59: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *p = request->getParam(0);
                                                           ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9545:59: note: candidates are:
In file included from C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:70:0:
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino: In lambda function:
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9564:59: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *p = request->getParam(0);
                                                           ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9564:59: note: candidates are:
In file included from C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:70:0:
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino: In lambda function:
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9626:67: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *slotParam = request->getParam(0);
                                                                   ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9626:67: note: candidates are:
In file included from C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:70:0:
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino: In lambda function:
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9673:72: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *slotIndexParam = request->getParam(0);
                                                                        ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9673:72: note: candidates are:
In file included from C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:70:0:
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9681:71: error: invalid conversion from 'const AsyncWebParameter*' to 'AsyncWebParameter*' [-fpermissive]
[... 10 lines cut ...]
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino: In lambda function:
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9806:58: error: call of overloaded 'getParam(int)' is ambiguous
                 request->send(SPIFFS, request->getParam(0)->value(), String(), true);
                                                          ^
C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:9806:58: note: candidates are:
In file included from C:\Users\tjspe\Downloads\gbs-control\gbs-control.ino:70:0:
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
c:\Users\tjspe\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
exit status 1

Compilation error: call of overloaded 'getParam(int)' is ambiguous

YoImLee, 2025-04-25:
Try this version of ESPAsyncWebServer (2.10.8): https://github.com/me-no-dev/ESPAsyncWebServer/tree/b70d36f2bec270e504bdc739b91a0fcbf3d84377

trev1976, 2025-04-25:
Hi , thanks. I will try that. I have a new D1 mini coming as well as I'm sure my one is playing up. :)

trev1976, 2025-04-29:
i just got my new d1 mini and this version uploaded. thanks

cpawliuk, 2025-07-26:
This is an easy fix, you just need to make sure you have **const** before AsyncWebParameter.

Anything with _request->getParam(1);_ should be **request->getParam((size_t)1);** Same applies if the number is not 1 or another index.

After you fix those you'll need to make an edit to both AsyncJson.h and AsyncMessagePack.h where you see _#if __has_include("ArduinoJson.h")_ you'll need to swap it out with **#ifdef ARDUINOJSON_VERSION_MAJOR**

**AsyncJson.h**
`#ifdef ARDUINOJSON_VERSION_MAJOR
  #include <ArduinoJson.h>
  #if ARDUINOJSON_VERSION_MAJOR >= 5
    #define ASYNC_JSON_SUPPORT 1
  #else
    #define ASYNC_JSON_SUPPORT 0
  #endif
#else
  #define ASYNC_JSON_SUPPORT 0
#endif`

**AsyncMessagePack.h**
`#ifdef ARDUINOJSON_VERSION_MAJOR
  #include <ArduinoJson.h>
  #if ARDUINOJSON_VERSION_MAJOR >= 6
    #define ASYNC_MSG_PACK_SUPPORT 1
  #else
    #define ASYNC_MSG_PACK_SUPPORT 0
  #endif // ARDUINOJSON_VERSION_MAJOR >= 6
#else
  #define ASYNC_MSG_PACK_SUPPORT 0
#endif`