ramapcsx2/gbs-control #614: No VGA Signal as soon as I connect the ESP8266 board? (Works fine when I disconnect it) [0.00000HZ output?]

BOBdotEXE, 2025-05-01:
Hello, I've been trying to build a GBS control, but I have this weird issue where my monitors can't see it as soon as I attach the board.
I've tried direct VGA and with and HDMI adaptor, both work fine on the same GBS board normally, (jumper removed)
But attaching the ESP8266  seems to cause it to not transmit a signal regardless of my jumper's position.

I've made sure to test with just the ESP8266 and no other mods, even without a jumper but my monitor still will not detect a signal.

I've also tried installing the clock mod an and oled display, both seem to function properly, and this is how I can see, despite having a working component device connected, and detected, it's reporting a 0.0HZ output?
[image]

I know I screwed up the removing the pots, but I properly wired to the resistors, and it works just fine with no esp connected:
[image]
(Sorry for the messy soldering, I'm still new, and have been messing around with this for hours..)

Same with the clock mod, no bridged connections, and if I removed the esp (and jumper), and leave the clock mod, the gbs still works fine:

[image]

[image]

I've also tried removing and resoldering the wires to the EPS, and even soldering directly to the smaller boar based on a pin out diagram,
still no luck:

[image]

As far as I'm aware I've double and triple checked everything, resoldered every point, and checked for bridged contents, but no luck!

The debug log did not seem to help much either:

```
startup
(WiFi): AP mode (SSID: gbscontrol, pass 'qqqqqqqq'): Access 'gbscontrol.local' in your browser
ext clockgen detected
<reset>
Chip ID: 1A 1
 G: 3E G: 3F G: 40
 R: 3E R: 3F R: 40 R: 41 R: 42
 B: 3E B: 3F B: 40 B: 41 B: 42
<reset>
Activity detected, input: Component

2345678
Format change: 1 <stable>
ADC offset: R:42 G:40 B:42
clock gen reset: 108000000
HTotal Adjust: -4

preset applied: 1280x720 for NTSC 60Hz 

sync skipped sfr wrong: 0.00

```
However from what I understand _"sync skipped sfr wrong: 0.00"_  means it can't detect the debug pin, so I have no idea what's going on.

[image]

I even followed all the troubleshooting guide steps, including reversing SDA / SCL.
(The Debug log for reversed shows "Please check board power and cabling or restart!" so I had it right the first time)

Any suggestions other than just "buy a new GBS and esp and try again?"

YoImLee, 2025-05-02:
It looks like you forgot to solder a wire from the CLK0 pad on the clock generator to pin40 of the chip below. Pin 40 is the outermost pin directly below. Have a look at the first picture here: https://github.com/ramapcsx2/gbs-control/wiki/Si5351-Clock-Generator-install-notes

YoImLee, 2025-05-02:
The clock generator pin is necessary to get an output unless you toggle the "disable external clock generator" option in the settings and restart the esp, but that would kind of defeat the purpose of having it installed in the first place.

BOBdotEXE, 2025-05-02:
> The clock generator pin is necessary to get an output unless you toggle the "disable external clock generator" option in the settings and restart the esp, but that would kind of defeat the purpose of having it installed in the first place.

Oh wow, yeah that seemed to do the trick! Thanks for the help and explanation!

[image]

[image]

I guess this means when I first installed the board, and flipped the jumper it was still expecting the clock generator by default?

Did I miss this in the instructions somewhere?, I don't remember reading that in the 'Basic Install' or 'troubleshooting'  steps?

closed 2025-05-02 by BOBdotEXE

YoImLee, 2025-05-02:
The ESP checks whether or not a clock generator is installed during startup. The clock generator board is connected via the serial lines and therefore detected, but because the wire for the clock output was missing the scaler chip can't output a picture on screen. 

If you were to remove the SCL/SDA wires to the clock generator board it would no longer be detected and the GBS falls back to using its own clock instead, screen tearing goodness included.