ramapcsx2/gbs-control #615: Web GUI Stopped Working

roddersuk, 2025-05-04:
I built my GBS-Control box a while back and it worked fine connecting my Amiga 500 to an LCD screen. The Web GUI worked fine and I was able to control the settings.
Recently I started work on an Acorn A3000 and wanted to use the GBS-Control to connect it up. I had problems getting a satisfactory image but when I tried to load the GBS-Control Web GUI I just got the screen of death with the flashing red WiFi icon.
I don't understand what could have changed that caused this but connection was a problem from my laptop, tablet or phone.
I've tried rebuilding the firmware again, trying to stick to the correct versions of libraries but to no avail.
I can ping gbscontrol on 192.168.4.1 but the GUI stubbornly refuses to cooperate.
I think that its failing to open a socket to the GBS-Control but I have no idea how to fix it.

YoImLee, 2025-05-06:
Hi, make sure you are using the correct version of ESP8266 (below 3.0, ideally 2.6.3 or so) and the ESPAsyncWebServer: https://github.com/me-no-dev/ESPAsyncWebServer/tree/b70d36f2bec270e504bdc739b91a0fcbf3d84377

Another thing to try is going into the Arduino IDE and selecting `Tools > Erase Flash: > Sketch + WiFi Settings OR All Flash Contents` before flashing to fully reset the ESP. 

It might also help to turn off whatever source is connected to the GBS when trying to access the web GUI, as I believe it's possible for the big chip on the board to cause interference.

roddersuk, 2025-05-07:
Thanks for the advice.
Unfortunately That is exactly what I've tried. I have those exact versions of code and I've been trying different versions of ESP8266TCP but it still won't work. I also erase the WiFi settings and try to connect to the GUI with no video signals.
I can connect to the gbscontrol AP with no problem and it pings fine on 192.168.4.1 but the GUI stubbornly refuses to work.
The curious part of this is that it _was_ working and when I tried to connect some time later it wouldn't so nothing had changed on the GBS-Control, I can only assume that some security update on the browsers on my phone/tablet/laptop has caused this issue.
I'm just surprised that it doesn't affect everyone.
Its turned the GBS-Control into something of a brick!

roddersuk, 2025-05-09:
I've found where the problem is. The javascript is failing in `fetchSlotNames`. 
Its getting a 500 error when calling `fetch(`/bin/slots.bin?${+new` Date()}`)``
I guess `http://gbscontrol/bin/slots.bin` doesn't exist. How should it get created?

roddersuk, 2025-05-17:
Well I found a solution. Trying to debug the code was not very fruitful (I'm still getting my head around promises!) but I surmised that there was something stored on the Wemos that was not being reset when uploading the sketch. I therefore wiped it using `esptool.py -c esp8266 -p /dev/ttyUSB0 erase_flash` and uploaded the sketch again and voila, it worked.
Why it had stopped working without any changes being made remains a mystery but at least I know how to fix it now.

DavidAndrewHeron, 2025-06-22:
> Well I found a solution. Trying to debug the code was not very fruitful (I'm still getting my head around promises!) but I surmised that there was something stored on the Wemos that was not being [...]

Just wanted to let you know that I had this EXACT same issue and this fixed it. I was able to ping and access my GBS-control/webpage for it, but the settings would never populate; it just had a red wifi icon blinking in the top right. This persisted across 4 different firmware versions and reflashes/restarts and it was even worse once I reset the wifi chip's settings because this was still present on the local webpage via the gbscontrol SSID.

After formatting and THEN reflashing the firmware, everything worked fine.

Thanks so much; you saved me a ton of frustration.