ramapcsx2/gbs-control #618: How do I upgrade my GBS-C's firmware to the latest?

Ahri-man, 2025-05-13:
I purchased one of these on 31, Aug, 2022:

[image]

How do I update its firmware? is there a .bin file and a flasher.exe that can do it? where do I get the latest .bin file? I am afraid I could brick the unit.

Many Thanks

BOBdotEXE, 2025-05-17:
> How do I update its firmware? is there a .bin file and a flasher.exe that can do it? where do I get the latest .bin file? I am afraid I could brick the unit.
> 
> Many Thanks

For pre-made models like that I'd always recommend contacting the seller/manufacture for details on how to update.

GBS control is designed to be compiles from source based on the instructions.

But the normal setup process requires connecting directly to the ESP88266 via USB:

[image]

if yours has a chip like that, and you have a way to connect to it,
 you may be able to follow the compile and upload instructions listed here:
https://ramapcsx2.github.io/gbs-control/Wiki/Software-Setup.html

But again, I'd recommend contacting the seller first if they made some changes, you don't want to 'flash' without a backup

Ahri-man, 2025-05-17:
I contacted the seller and he said that there are no firmware updates for this.

Is there a danger in trying to upgrade the firmware? could i possibly brick the unit?

BOBdotEXE, 2025-05-19:
> I contacted the seller and he said that there are no firmware updates for this.
> 
> Is there a danger in trying to upgrade the firmware? could i possibly brick the unit?

You could ask the seller if it's compatible with ramapcsx2's firmware, but it's starting to sound like the answer is no.

And as a general rule, with devices that run modified versions of open source projects, if you don't have a backup of their specific firmware, I would not recommend trying to flash anything else.

If you did manage to get the official GBSc firmware flashed on this, and it fails to boot. (or dosent work properly) without a copy of compatible firmware,  your device would pretty much be bricked, with no way to restore it's functionality.