ramapcsx2/gbs-control #625: Deinterlacing is not working on Amiga

VelosCohagen, 2025-05-28:
Hello!

Im using GBS 8200 with my Amiga 500 with 640x512 interlaced resolution. Amiga is connected via RGB to VGA Adapter with CSync.
Newest GBSC firmware is installed on the device.

The image on the desktop is really flickery, so deinterlacing doesn't seem to work.

If further information is required, please reply. I'll try to deliver the needed informations.

ProbDenis, 2025-05-29:
Bob doesn't really deinteralce, so the flickering is expected here. Motion adaptive works, but it does a very poor job when the screen is mostly static (productivity software, workbench, strategy games etc.) and causes visible artifacts around the mouse cursor (see #287).

The solution would be to have simple weave deinterlacing as an option, as described in #336. That should result in a much more stable image for these use cases.

VelosCohagen, 2025-05-29:
Is there a special cable or adapter i should try? Maybe this is the problem.

VelosCohagen, 2025-05-29:
@ramapcsx2 Thanks for your reply. As stated, i'm using 640x512i 50Hz PAL on Amiga OS desktop.

jojo-gfx, 2025-06-29:
Here the same problem.
Input from Amiga per VGA port views no flickerfixer function and also changing to Bob shows absolute no change.

Have test it on my ordinary GBS without GBS Control and there flickerfixer is working.

The difference is, I'm using on the old one the input pins behind the VGA port. If I using the VGA port, like on the GBS Control, no picture is shown.

Could it be, the VGA port isn't ready for Amiga Signal?