ramapcsx2/gbs-control #627: question

AlfredoDore, 2025-06-07:
I'm going to buy a GBS-C and I have some questions that I'd like to clear up since I haven't seen anyone talking about it, so here they go:
1 - I have an LCD monitor that has native analog inputs that include: VF, AV, S-video, Component, VGA and a digital HDMI, my intention is to create an experience as close as possible to a CRT, so I think that if I connect the GBS-C to the analog outputs of my monitor, I could create an experience close to scanlines, like the scanlines of the GBS-C?
2 - I'm aiming at one that looks like its RCA output is from the Sega Saturn, could you clear up this doubt for me before I buy a cable and end up buying the wrong cable?
I intend to buy this one, here are its video outputs, as you can see it has three, RCA, VGA and HDMI, would the RCA be the Sega Saturn cable?
[image]

PabloNewt, 2025-06-12:
For 1) In my opinion the purpose of GBSC in this scenario is to give you more control over how the video looks on your LCD monitor. If you use its analogue inputs it sort of defeats the purpose. Your monitor probably has similar hardware internally to GBSC but with less control over the result. I think the analogue output of GBSC is more for people using CRT monitors that aren't natively compatible with their console.
For 2) In the image above it has "SEGA Saturn" written on the PCB above the "DIN" socket (it's not RCA) so it is definitely the right one.

AlfredoDore, 2025-07-06:
> For 1) In my opinion the purpose of GBSC in this scenario is to give you more control over how the video looks on your LCD monitor. If you use its analogue inputs it sort of defeats the purpose. [...]

I bought it, it hasn't arrived yet, so, the Din 10 output, commonly associated with the Sega Saturn, does it take a composite cable? (that yellow, red and white cable)