ramapcsx2/gbs-control #633: GUI not loading

rabidgerry, 2025-07-16:
Hi guys, I have been using the GBS with GBS control for some years now and never encountered this issue.  Basically I can connect to the GBS but the GUI is not opening.  It tries to be that's about it.  You can see what I mean in this video.  I have tried the most recent firmware also and still I cannot open the GUI.

https://youtu.be/0KiHKrJcF1s

I also tried to have the GBS forget my wifi settings but this did nothing also so I'm back to having to select it on my phone through the newtwork options.  If anyone has any ideas I'd be very grateful.  As it is I'm stuck with the current setting as I can't access the gui.

rabidgerry, 2025-07-20:
Managed to get it working by erasing everything totally on the board using the arduino software and "erase all flash contents" option.  So I can now connect via my phone.  However I am unable to add the gbs to my wifi network despite reserving an IP for it.  When I add in the wifi password and hit connect, the board resets as it's supposed to and then nothing.  Anyone have any ideas to try to solve this issue?  It used to accept my wifi details no problem so I do not know what's up?

trickydee, 2025-09-24:
I had the same issue, with the the UI not loading with the browser client constantly trying to load /slots.bin?<number> .
I am using a D1 pro mini with an external antenna specifically to improve the wireless connectivity. And the GBS is very close to the access point.

I tried erasing the flash using "erase all flash contents" in the Arduino IDE as suggested but this did not work. I also tried ESP tool neither worked: 

esptool -c esp8266 -p /dev/cu.usbserial-xxxxxx erase_flash.     .

But then I tried changing the Arduino IDE Flash_Size setting to "Defined by Hardware and Sketch", I then refreshed and everything worked as expected !.

So as with other folk with this issue it looks to be caused by some kind of corruption and no slot config being available initially. I wonder if this could be fixed by having default slot definitions added to the code?