ramapcsx2/gbs-control #635: unable to compile due to errors

alienkidmj12, 2025-07-25:
using the arduino tool im getting the following when trying to compile code

/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino: In lambda function:
gbs-control:9545:59: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *p = request->getParam(0);
                                                           ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:9545:59: note: candidates are:
In file included from /home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:70:0:
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino: In lambda function:
gbs-control:9564:59: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *p = request->getParam(0);
                                                           ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:9564:59: note: candidates are:
In file included from /home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:70:0:
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino: In lambda function:
gbs-control:9626:67: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *slotParam = request->getParam(0);
                                                                   ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:9626:67: note: candidates are:
In file included from /home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:70:0:
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino: In lambda function:
gbs-control:9673:72: error: call of overloaded 'getParam(int)' is ambiguous
                 AsyncWebParameter *slotIndexParam = request->getParam(0);
                                                                        ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:9673:72: note: candidates are:
In file included from /home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:70:0:
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
gbs-control:9681:71: error: invalid conversion from 'const AsyncWebParameter*' to 'AsyncWebParameter*' [-fpermissive]
                 AsyncWebParameter *slotNameParam = request->getParam(1);
                                                                       ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino: In lambda function:
gbs-control:9711:51: error: call of overloaded 'getParam(int)' is ambiguous
         AsyncWebParameter *p = request->getParam(0);
                                                   ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:9711:51: note: candidates are:
[... 5 lines cut ...]
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino: In lambda function:
gbs-control:9806:58: error: call of overloaded 'getParam(int)' is ambiguous
                 request->send(SPIFFS, request->getParam(0)->value(), String(), true);
                                                          ^
/home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:9806:58: note: candidates are:
In file included from /home/sivesrutherfordd/Arduino/libraries/gbs-control/gbs-control.ino:70:0:
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:403:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const char*, bool, bool) const
     const AsyncWebParameter* getParam(const char* name, bool post = false, bool file = false) const;
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:405:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(const String&, bool, bool) const
     const AsyncWebParameter* getParam(const String& name, bool post = false, bool file = false) const { return getParam(name.c_str(), post, file); };
                              ^
/home/sivesrutherfordd/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:416:30: note: const AsyncWebParameter* AsyncWebServerRequest::getParam(size_t) const
     const AsyncWebParameter* getParam(size_t num) const;
                              ^
Multiple libraries were found for "SSD1306Wire.h"
 Used: /home/sivesrutherfordd/Arduino/libraries/ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays
 Not used: /home/sivesrutherfordd/Arduino/libraries/esp8266-oled-ssd1306-master
exit status 1
call of overloaded 'getParam(int)' is ambiguous

cpawliuk, 2025-07-26:
See my reply to this here: https://github.com/ramapcsx2/gbs-control/issues/613#issuecomment-3121583315

Same solution, I was able to compile just fine after those changes. If it still wont work let me know and I'll walk you through it.

alienkidmj12, 2025-07-27:
you mentioned this "This is an easy fix, you just need to make sure you have const before AsyncWebParameter.

Anything with request->getParam(1); should be request->getParam((size_t)1); Same applies if the number is not 1 or another index."

which files am i looking in for this ?

cpawliuk, 2025-07-27:
If you still have any more problems you can reference the repo I uploaded with the corrections made here: https://github.com/cpawliuk/gbs-control-complete

cpawliuk, 2025-07-28:
I'm not sure, but you'll get an error log anyways when you try to compile the code if there are any issues. If you're not sure what changes you need to make just post back and I'm sure someone will help.

However, from what I can see the compiler is requiring explicit type casting to `size_t` - so for `getParam(1)` you need it to be `getParam((size_t)1)` because `getParam` is an overloaded method, also due to implicit conversions.

The other error where you need to go from `AsyncWebParameter *p = request->getParam((size_t)0);` to `const AsyncWebParameter *p = request->getParam((size_t)0);` is just const correctness.