ramapcsx2/gbs-control #636: Aspect ratio is not 4:3

markandgo, 2025-07-27:
I recently purchased a pre-built GBS-C with HDMI output. I am using component cables with my PS2. With the 1920x1080 preset, the image should scale up and preserve its 4:3 aspect ratio, if my understanding is correct. I noticed that many games do not seem to stretch up to 4:3 correctly, because some games look to have a wider or narrower image than others.

ramapcsx2, 2025-07-28:
Yea, there were several hard to do aspects of upscaling, and if pressed, I would prefer image quality over aspect ratio concerns, within reason.
The HD output preset is most affected by it, too. You can try improving it on the output device settings, if it has some.
The scaler itself though.. what you get is the best it can do :p

markandgo, 2025-07-29:
I'm going to close my issue as I had a misunderstanding of how CRTs worked. It seems that for an NTSC signal, a CRT will always display 480 lines, but not all games used 480 lines to draw. That explains why some PS2 games had aspect ratios that were not 4:3 because of the different vertical resolutions. As for horizontal resolutions, it seems GBS-C correctly upscales horizontally with 1280x960 so that all games will have the same horizontal resolution or "length" just like how a real CRT work. Thank you for your work!