ramapcsx2/gbs-control #637: VGA output

AlfredoDore, 2025-07-30:
I bought the GBS-C and it arrived today, but I'm having problems with the VGA output, it's acting like there's no output, it's my first GBS-C and I'm still learning how to use it, mine is this one: https://pt.aliexpress.com/item/1005008835546610.html?time_tag=462688777277452&aff_fcid=c6bb832cda9e4aca93879b8f0611cc04-1753882012192-05995-_oBwTAWR&aff_fsk=_oBwTAWR&aff_platform=api-new-link-generate&sk=_oBwTAWR&aff_trace_key=c6bb832cda9e4aca93879b8f0611cc04-1753882012192-05995-_oBwTAWR&terminal_id=a4ed0b4f0a994436b112a497cecea7e4&afSmartRedirect=y

cpawliuk, 2025-07-30:
I would do a few tests assuming you know your input source is fine.

1. See if HDMI output at least works on the unit to at least factor out the GBS-C not outputting anything.
2. If HDMI works fine then check your VGA solder joints, also check your cable and input source (Monitor / TV) - maybe try different resolutions. Try other cables, other screens if possible.

Are you using the latest firmware as well? A lot of these use very dated or custom firmware based off old builds.

With these pre-built units it is always a good idea to fully inspect the board if possible. I've seen "questionable" build quality on these, and it might require some soldering to fix bad joints.

Posting a video would help as well to see what is turning on. You also view the LOG from the Web GUI.

AlfredoDore, 2025-07-31:
> I would do a few tests assuming you know your input source is fine.
>
> 1. See if HDMI output at least works on the unit to at least factor out the GBS-C not outputting anything.
> 2. If HDMI works fine [...]

 I did it, now I'm going to see how I update the firmware, I really liked the result of the scanlines, but I want to see what features there are in the newest firmware

[image]